teachingtechYT/teachingtechYT.github.io #340: Checking 'Strip ALL original start gcode and use only custom gcode instead' adds 'undefined' Symbol in gcode-file

KoeRt888, 2021-08-01:
Hi,

I'm using Klipper and have a start code macro to let the bed heat up for 5 minutes to wait for the expansion of the aluminum bed. When I try to use it as only means as start gcode the generator on the 'First Layer' calibration page it adds a 'undefined' symbol to the gcode-file. I don't know if this happens in every calibration page form or just the one of 'First Layer'.

First try to copy my start code macros:
[image]

Try to copy it from notepad++ to the field with an additional CR LF:
[image]

result in the downloaded gcode:
[image]

Is this a symbol that is added via javascript, or is has it something to do with encoding of the string pasted to the field from Slicer software (SuperSlicer) to browser?

teachingtechYT, 2021-08-01:
That is a bug specific to the first layer test that I've been unable to find the cause of. It happens when a variable is refrencing in the gcode procesing that does exist or can't be accessed.
It's been there since the V2 rewrite but typically was tacked on to the end of a comment so was never read by the machine. Using the strip start gcode seems to expose on its own line. Thanks for the report, I'll do some digging.

teachingtechYT, 2021-08-01:
Fixed!

closed 2021-08-01 by teachingtechYT