teachingtechYT/teachingtechYT.github.io #347: Suggestion: Flow Calibration Instruction/Warning

Brucien, 2021-08-07:
It may be worth noting (as I had to figure it out myself) to make sure "alternate extra wall" is ticked off in Cura, as well as whatever the equivalent is in other slicers. While it's likely this will not affect people new to printing, people like myself looking to re-tune their machine may have it on in there profile.

teachingtechYT, 2021-08-08:
I'm not familiar with this feature. Would you mind elaborating?

Brucien, 2021-08-08:
Sure, it is an option that adds an extra wall on the inner perimeter on every other layer. This is off by default and hidden under the basic view in Cura. The purpose is to increase the strength of the bond between the shell and infill. When printing the cube, even though one wall and 0% infill was selected, the alternating extra wall still printed on the inside, giving me a measurement of .7mm on my walls. It only took a few minutes to figure out why that happened, and was not a big deal, I just thought that a brief mention could keep someone from stuck.

teachingtechYT, 2021-08-08:
Thanks for explaining. I agree it's worth mentioning. Tab updated.

closed 2021-08-08 by teachingtechYT