teachingtechYT/teachingtechYT.github.io #349: Curling on Ender 3 v2

OlSom, 2021-08-11:
I put together my Ender 3 v2 a few days ago, so I'm fairly new to all of this. After tuning e-step I wanted to print the temp calibration, but every time I tried to do so, the left side of the print on the first arches curled up so much that after a few layers the nozzle hit them and dislodged the print from the bed. I tried a bunch of different temperature settings down to 190, 0.16 layer height as well as slower speeds down to 45mm/s but all with the same result:

[image]

While it clearly seems related to the cooling, I also tried slicing the .stl in Cura, with an Ender 3 profile that I have used before for a test cube and a benchy, and that printed pretty well:
[image]

The generated gcode from the site:
https://pastebin.com/gdibpRyN

My gcode from Cura:
https://pastebin.com/YJdWZw85

I'm too new to this to find the meaningful differences in the code.

teachingtechYT, 2021-08-15:
I've had a look at the two sets and I can't see anything significant. Both are doing the perimeters from the inside out, same infill. Site has fan on 100% non stop whereas Cura varies it a bit.
Speeds might be where it's at. Cura is slowing down layer based to hit a minimum time. You can see the exterior walls are much slower lower down (~10mm/sec) compared to the bridge (~25mm/sec):
[image]
Site gcode is not only faster but consistently faster throughout (~30mm/sec):
[image]
My guess is part cooling is the limitation, and the Cura profile is slowly everything down to mask that, but the site is not. You would need to lower the feedrate for the site a great deal more to match what Cura is producing. Even though your default speed in Cura is no where this low, the modifiers for perimeters make it lower, and then the minimum layer time is making it even slower.

OlSom, 2021-08-18:
Lowering the feedrate on the site to 30 did indeed fix curling and I could print the piece and get a good result for the different temperatures. Time to look into cooling updates I guess.

Thank you for the help!

closed 2021-08-18 by OlSom