teachingtechYT/teachingtechYT.github.io #350: z offset is different between cura and calibration gcode

cdeitrich00, 2021-08-14:
weird issue... when i run prints from cura my z offset is perfect but if i run any of the calibration prints my nozzle is a bit too far from the bed

i noticed these lines in the gcode 

G28 ; home all axes
G29 ; probe ABL
;M420 S1 ; restore ABL mesh 
;customstart
G0 Z3; fix for delta printers that home at max
; square 1
;process Process
;layer 1, Z = 0.200

maybe G0 Z3 is part of the issue im not great with gcode so im not sure

my z-offset in cura is set to 0 so i dont think cura is compensating

teachingtechYT, 2021-08-14:
There's nothing my my gcode to change the Z offset. That would be an M851 command.
The G0 Z3 line just moves Z to 3mm above the bed before starting the print, which is required for deltas which home at the top. If you keep reading the following gcode should run at Z0.2 or whatever layer height you selected.
I would suggest comparing the first printing lines from the gcode on the site vs the first printing lines from your slicer to look for a different Z height.
For example, Cura might be printing the first layer at a higher thivkness compared to this site which keeps the same layer height the whole way through.

teachingtechYT, 2021-08-23:
No response, closing.

closed 2021-08-23 by teachingtechYT