teachingtechYT/teachingtechYT.github.io #351: BL touch on 4.2 boards

Ewollin1978, 2021-08-14:
When using the dedicated port on 4.2 boards   

if your trigger is plugged into a dedicated probe port you must make sure Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is disabled 
eg //#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
and enable USE_PROBE_FOR_Z_HOMING (if you don't have a real Z endstop and are using the probe) 
eg #define USE_PROBE_FOR_Z_HOMING

I did not see instructions for this on the guide but found on a reprap forum post by user Dust. Since I was installing a CR touch this was the only option.

teachingtechYT, 2021-08-14:
Forgive me if I've misunderstood, but aren't you talking about what's already on the page:
https://teachingtechyt.github.io/upgrades.html#bltouch
[image]
Please reopen and explain if this is not the case.

closed 2021-08-14 by teachingtechYT

Ewollin1978, 2021-08-16:
I think the third line threw me off in the slide as I was using a CR touch and the pin out was default I did not need to set Z_Min_Probe_Pin. Keep up the great work a lot of what I have been doing would have been way over my head if not for your site. One suggestion as it was difficult for me to follow along in your step by step instructions ( I am a total newb) the pace of the video is very fast for someone like me and by the time I hit the pause button on the screen you were two steps ahead of me. It might be help full for us Newbs if you could break them in to short videos of each steps. You could use the same videos just chop them. 
So far I have upgraded my hot and added direct drive, upgraded my main board to a 4.2.7 added Octo Print  now a CR touch and Octo Laps all with your help. I would have never had the confidence to do that with out your assistance, I really appreciate what you are doing here. I am super interested in setting up linear advance now but still a little on the fence as to my ability.
Thanks again

https://teachingtechyt.github.io/upgrades.html#bltouch