teachingtechYT/teachingtechYT.github.io #352: Using mathematical optimization for tuning retraction/temp.

cmdremily, 2021-08-15:
I think we can do better in terms of process for calibrations in general but for retraction, or reducing stringing in particular. Currently the process is, try changing things one at a time and to do a truck-ton of test prints. This is equivalent to doing a semi-random line search process in optimization theory. Convergence of such a process is slow with many steps.

With appropriate tool support (TT calibration site) and suitable test print design and judging conditions we should be able to (at least approximately) apply the principles of mathematical optimization techniques in particular gradient descent or grid search. 

I'm outlining a proposal for a process for grid search below:

1) Pick a starting point. E.g. your current retraction/temp settings or default settings from your slicer.
2) Print a test piece that varies each of the 6 variables (retraction speed/distance, reset speed/distance, z-hop and nozzle temperature) by two steps in either direction with user chosen step size. 
3) Input the best result from each variable into the tool.
4) Tool picks a new starting point from the best values, and presents a new test print with possibly smaller step size.
5) Repeat from 2 until no notable improvement is found

As long as the step size is chosen appropriately the above method is guaranteed to eventually converge on a local minima for stringing. If the stringing as a function over the parameters is convex, then the local minima is also the global minima. 

The above presents a structured and guided approach for correctly minimizing stringing under by-conditions of how much you're willing to deviate each value from their starting position. E.g. don't vary temperature by setting step size to 0. It would be very cool if the TT calibration site would consider implementing support for such a guided and process.

Keep up the good work!

(Note: Although possible, applying gradient descent would be more complicated as we'd need to estimate the gradient from a test print, and that would involve objectively estimating how much better stringing got on each variable we changed. This is subject to large errors in objective judging and hard to do compared to picking the best out of N tests as we do above).

(Note 2: The user would ideally first print a temperature tower, and based on desired strength, overhang and bridging parameters pick a temperature range they are comfortable with, and then proceed to tuning retraction to remove stringing within that chosen range.)

cmdremily, 2021-08-16:
Actually, now that it's not midnight, it's not quite grid-search... It's a form of [pattern search](https://en.wikipedia.org/wiki/Pattern_search_(optimization)) that I meant.

teachingtechYT, 2021-08-16:
It's clear that you understand this very well but to most visitors to the site their head is goiung to spin. To implement such a methodology it needs to be communicated in a simple way, using text plus diagrams and examples yet remain concise.
The site is already very text heavy and dense, so I am willing to add more instruction but it must conform to the criteria I've listed.

closed 2021-09-01 by teachingtechYT