teachingtechYT/teachingtechYT.github.io #354: firstlayer.gcode is "empty" for 0.50 mm nozzle / 0.35 mm layer height

BorisUK, 2021-08-21:
Steps to reproduce:

1. Go to [First layer page](https://teachingtechyt.github.io/calibration.html#firstlayer)
2. Select nozzle diameter / layer height: 0.50 mm / 0.35 mm
3. Press Download Gcode

The firstlayer.gcode that gets generated is virtually empty (754 bytes). Squares 1-5 only contain "undefined" text. Below is the full gcode file, with all the other settings. Other nozzle diameters or layer heights I tried produce 20k files.

gcode file contents:

; G-Code originally generated by Simplify3D(R) Version 4.1.2
; This calibration test gcode modified by the Teaching Tech Calibration website: https://teachingtechyt.github.io/calibration.html
;M80 ; power supply on
G90
M82
M106 S0
M140 S60 ; custom bed temp
M190 S60 ; custom bed temp
M104 S210 T0 ; custom hot end temp
M109 S210 T0 ; custom hot end temp
G28 ; home all axes
;G29 ; probe ABL
;M420 S1 ; restore ABL mesh 
;customstart
G0 Z3; fix for delta printers that home at max
; square 1undefined
; square 2undefined
; square 3undefined
; square 4undefined
; square 5undefined
G28 X0 ; home X axis
M106 S0 ; turn off cooling fan
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M501 ; restore previous EEPROM values
;customend

teachingtechYT, 2021-08-23:
Hi Boris,
You have indeed found a bug. It appears I exported that one at 0.3mm layer height instead of 0.35mm.
The site is looking for [50_35] but only [50_30] is there and it comes up blank.
I will reexport and update this week, sorry for the inconvenience.

teachingtechYT, 2021-08-31:
Fixed.

closed 2021-08-31 by teachingtechYT