teachingtechYT/teachingtechYT.github.io #359: Klipper g-code support

Critterninja, 2021-08-29:
Would like to request support for Klipper g-code generation support.  Currently, I get 'move exceeds maximum extrusion' errors.  even adding in "max_extrude_cross_section:" into the config, it still way over extrudes and errors out.  This seems to be an issue with S3D and Klipper, which I believe is used for the backend gcode generation.

teachingtechYT, 2021-08-31:
Can you be more specific on whicg print this occurs and at what portion?
I have been using S3D successfully for Klipper gcode with two printers and haven't come up against this yet.

Critterninja, 2021-08-31:
Attached is the baseline code generated on the site that I used.  The start
code is a copy paste from my slicer that works fine and covers everything
up to the priming.  The error happens just after it primes and then goes to
start the print.  It never actually starts, but kicks out an error.  below
is the first error

Move exceeds maximum extrusion (1.422mm^2 vs 0.640mm^2)
See the 'max_extrude_cross_section' config option for details

After this, it did it again.  After looking this up, I found the posts
stating that adding, and then increasing the max_extrude_cross_section in
the config to around 50 helped solve this issue when it was a problem.  And
the common denominator between posts referenced S3D as the slicer that
showed this issue.  But upon applying this settings, it printed half of the
first layer and then errored out again with the below

Move exceeds maximum extrusion (50.122mm^2 vs 50.000mm^2)
See the 'max_extrude_cross_section' config option for details

I tried to apply the settings as best as I could in PrusaSlicer to match
what you were going for on the site, and it printed fine.  Also works in
SuperSlicer too, tho it is a fork of PrusaSlicer to be fair.  I have not
tested in Cura.

Should also be noted that I'm using Klipper
version release-210618-1-g9aaa2d91, which is the Desuuuu version that
supports the touch display on my Ender 5 Plus, FLuidd v1.16.2, and
moonraker v0.7.1-42-g81176a6, if that helps any

Critterninja, 2021-08-31:
Figured I should upload a test cube I made directly in PrusaSlicer that worked fine for you to compare.  I just generated a new gcode on the site and ran it again, just in case I missed something and it stopped again.  tbh, it looks like it's the extrusion that going wonkers.  The prime line looks perfect, but when it started to print the skirt and then cube, it was pushing out a ton of filament like it did yesterday.

[TestCube_Prusa_0.2mm_PLA_14m.gcode.zip](https://github.com/teachingtechYT/teachingtechYT.github.io/files/7080608/TestCube_Prusa_0.2mm_PLA_14m.gcode.zip)

Critterninja, 2021-08-31:
another update.. more info the merrier.. 

I decided to generate the gcode on the site without injecting my start gcode and just using whatever your site generated.  It's actually printing fine.  I have no idea what my start code, that works fine in PrusaSlicer, has that yours does not, or vice versa.  other than Prusa is generating the code based on my firmware choice and compensates for the extrusion difference.  idk. I'm just guessing at this point because nothing stands out here.

I think the main reason I went with adding my start gcode was I wanted it to pull the saved mesh on my machine, verses have it do a probe, so it would save a few minutes of time.  Maybe have an option to pull the 'default' mesh using the bed_mesh_profile load="default" command. Worse case, if someone has custom profiles made, they can always do a quick mesh on the default profile and save it, then choose that option.

teachingtechYT, 2021-08-31:
Thanks for your detailed follow up. The problem is fortunately simple!
Your start gcode configures the machine for relative extrusion moves with M83:
[image]
Whereas the site uses absolute with M82:
[image]
Basically, the custom gcode you are using tells the macjine to expect one type opf command but then the rest of the gcode to to suit the opposite. Fortunately nothing to do with the type of firmware.
All you should need o fix is changing the M83 line to M82 in the cistom hcode you enter on this site. Your regular slicer start gcode stays the same.

Critterninja, 2021-08-31:
Just redid the gcode on the site using your suggestion and it worked like a champ!  More reasons why I love your content.  I'm always learning something new :)  Both the cubes that came out, using both your start code and my custom code, came out amazing.  I do appreciate all your help man!

teachingtechYT, 2021-08-31:
No problem! I'm just glad there wasn't some underlying issues with Klipper I had created.

closed 2021-08-31 by teachingtechYT