teachingtechYT/teachingtechYT.github.io #362: Firefox Start Gcode

JRosello, 2021-09-03:
Firefox 91.0.2 not show the _Additional Gcode Start_ box when the box is selected.
Tested in Chrome and it works so some issue with the page code with Firefox

teachingtechYT, 2021-09-06:
I just installed 91.0.2 Firefox and I am unable to replicate:
[image]
No errors in console either. I'm unable to shed any light at this stage but at least you have Chrome as a workaround.

JRosello, 2021-09-06:
Find the issue, I have installed an extension called "EFF Privacy Badger" to block trackers, so seems that if I had enabled for your site prevents from opening the gcode box... weird

teachingtechYT, 2021-09-06:
Glad you sorted it. This site doesn't use tracking cookies of any kind so that's very strange. The closest it does is store the values from the forms in the browser webstorage.

closed 2021-09-06 by teachingtechYT