teachingtechYT/teachingtechYT.github.io #368: Restoring EEPROM (used in retraction tower gCode) causes unexpected results

edwardhorsford, 2021-09-11:
Preamble: thank you very much for a helpful and informative resource.

----

I've had a really confusing and inconsistent time whilst trying the [retraction towers](https://teachingtechyt.github.io/calibration.html#retraction). Some prints would be fine, some would be massively under-extruded and have z-height issues.

I've finally worked out that it's caused by the `M501` reset to EEPROM at the end of the file - which I think might want to be reconsidered.

Issues:

* Running M501 at the end *only* means that the first print using this gCode will have whatever the printer's previous settings were, whilst subsequent prints will have EEPROM defaults. Thus two prints in a row can produce very different results.
* On the Prusa Mini, M501 resets the z-height (! [reported here](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1584)). I suspect this may be the cause of #350

## How this impacted me:
I've got a Bondtech extruder on a Prusa Mini so have custom E-Steps. You can have these set via start gCode, or via gCode you run when you turn on the printer. I've got a gCode file on usb key that I run when I start the printer.

I didn't expect that this gCode would reset my printer, so didn't add those extra eSteps in the start gCode section. With this gCode resetting the printer, the first time you run it, it has my correct e-steps. Thereafter it's reverted to the Prusa Mini default - which means I massively under-extrude. It took me forever (and many prints) to work out why some prints looked ok, and some looked terrible - all due to whether this was the second print since restart.

## Possible solutions
* Remove M501 from all files.

If removing M501 is not possible:
* Add to start of files?
* Add note / warning to page that the gcodes will reset EEPROM and so users may need to provide those *always*
* Add note / warning that gCode may reset z-height on Prusa Mini (will see what Prusa say about bug report)

## Next steps
I'll have a go providing my own end gCode and avoid M501. FWIW at the end whatever homing it does seems to do a miscalculation - it tries to move the Y carriage way beyond the correct position.

teachingtechYT, 2021-09-13:
Thanks for articulating this so clearly. I think the best course of action is to omit it for all tests.
It was added globally to suit the acceleration test where max accel valurs are modified in the test. I'll just add a note on that page to warn the user.

teachingtechYT, 2021-09-13:
Updated. Thank again.

closed 2021-09-13 by teachingtechYT