teachingtechYT/teachingtechYT.github.io #369: Temp tower fails printing every time.

Hawkeye666, 2021-09-19:
Michael, I hate to bother you but I've been trying to tune my Ender 5 Pro since I put the Micro Swiss direct drive and all metal hot end on. All the previous steps went swimmingly, but I cannot get past the temperature tower. I'm printing with PLA+ and the base prints fine, really, really well actually. But as soon as the layer with the bottom layer of the tower starts to print I lose layer adhesion. I tried setting the part fan to 100% at layer three but that didn't seem to help.

I suspect the issue here is something I am doing and not an issue with the process or the gcode, but this is esoteric enough for me that I'm not entirely sure about that. Any help would be greatly appreciated.

Cheers and thanks for all your work.

teachingtechYT, 2021-09-19:
I think this might be the same problem as this one: https://github.com/teachingtechYT/teachingtechYT.github.io/issues/349

Have a read at the solution and let me know if it works for you.

That page might need a warning about standard Cura profile settings vs the site.

Hawkeye666, 2021-09-20:
Thanks for the response. I will give this a shot and report back.

Cheers

Hawkeye666, 2021-09-20:
I made two changes to the settings and so far that seems to have done the trick. I dropped the base speed to 40, and I set the first test temp to 200, the same as the first layer temp, then increasing by 5 for each subsequent level. I have both a Micro Swiss all metal hot end and a hardened steel nozzle in place and I expect those two things combined to  make the starting temp of 190 too low for good adhesion for those very small towers.

Thanks for the help....

Hawkeye666, 2021-09-20:
So far I've tried lowering the feed rate to 40 and then to 30, in addition raising the first test's tempt to 200 and then 210. Both times the left side "tower," the smaller leg of the test object, gets knocked lose by the print head as it gets close to the bridging part of that first test. I think it might be curling at the corners that are getting caught by the nozzle and causing that left side to come lose.

I'm now going to try 30mm feed rate but starting at 230 and then lowering the tempt each test layer to see if that helps.

teachingtechYT, 2021-09-21:
No problem. At some stage I want to add a choice between the old and new test as the old one is less demanding on the printer.
In the meantime I'll add a note regarding speeds.