teachingtechYT/teachingtechYT.github.io #371: Invalid html code

cyberman, 2021-10-17:
Hello, 

not sure if this kind of feedback is wanted. 

I'm coming from the web design corner, and I'm using your project at my website. So I have a special focus for valid HTML 5. 

And I've found some issues which makes html code invalid - here are someone

dynamic.js, line 12

both target attributes of links have a wrong typo, must be "_blank" (not "blank")

calibration.html, line 32

target attribute of link has a wrong typo, must be "_blank" (not "blank")

calibration.html, line 738

target attribute of link has a wrong typo, must be "_blank" (not "_target")

calibration.html, line 413 and 420

missed typo, must be <thead> and </thead> (not <head> and </head>)

calibration.html, line 669 and 676

missed typo, must be <thead> and </thead> (not <head> and </head>)
 
Hope it's useful ;-).

cu Andy

cyberman, 2021-10-17:
I'm not familiar with GitHub yet. But I've tried now to put this issues in some commits.

teachingtechYT, 2021-10-17:
I think I got them all. Thanks very very.

closed 2021-10-17 by teachingtechYT