teachingtechYT/teachingtechYT.github.io #374: Support 0,15mm nozzle

TheDevilOnLine, 2021-11-12:
Currently the tooling support the 0,2mm nozzle as smallest nozzle, so I can't use it to calibrate my 0,15mm nozzle.

teachingtechYT, 2021-11-13:
You might be the only person around using one but I'm still willing to add support. What is your preferred layer height?

TheDevilOnLine, 2021-11-13:
Someone has got to be the first ;-).

I'm printing with 0.08mm and 0.12mm layer heights. If I'd have to pick one, I'd prefer the 0.08.

Thanks for your troubles!

teachingtechYT, 2021-11-13:
I think I would perfer 0.12 because it's aboce the rule of thumb of not exceeding 75% of the width of the nozzle diameter.
I'll try and add 0.12 this week.

teachingtechYT, 2021-11-15:
Realised I wrote thew opposite of what I intended in the last comment, which is now edited.
0.15mm nozzle diamter at 0.08 layer height now added for all tests.
Please let me know if you encounter problems.

closed 2021-11-15 by teachingtechYT