teachingtechYT/teachingtechYT.github.io #375: Acceleration Calibration overwrites Z axis Acceleration

flyingcadet, 2021-11-14:
My z-axis loses steps with z-axis accelerations over 10mm/s^2.  The GCODE generated overwrote with a "Z50000"...Yikes

Recommend eliminating Z-axis overwrite from all generated GCODE.  

See:  M201 X50000 Y50000 Z50000; custom raise acceleration limits

teachingtechYT, 2021-11-14:
For a cartesian printer this is an easy fix but the trouble is a delta needs X, Y and Z raised. Maybe the best solution is a check box in that section of the form for delta uses, so that non delta users only have X and Y raised for the test?

flyingcadet, 2021-11-14:
I wasn't aware of the acceleration cross coupling with delta printers.

flyingcadet, 2021-11-14:
Oops, wrong button on mobile.  Didn't mean to close the  issue.

teachingtechYT, 2021-11-15:
This should now be fixed.
A delta check box has been added just to the acceleration test. When checked, the M201 command is inserted with a Z value included, and if unchecked, M201 only raises X and Y.
Junction devaition values for Z are either included or omitted based on whether the box is checked as well.

closed 2021-11-15 by teachingtechYT