teachingtechYT/teachingtechYT.github.io #376: First Layer should raise nozzle between squares

micromouseonline, 2021-11-19:
Can I suggest that the nozzle be raised by a couple of millimeters while traversing between squares in this test? As it is, the nozzle stays at printing height and drags across the square. Not a problem as such, just a neatness feature.

For example, I modified the code like this for the first square:
```
G1 Z2.000 F1200
G1 X14.6360 Y16.3140 F5000 ; custom feedrate
G1 Z0.200 F1200
```

teachingtechYT, 2021-11-20:
Z hop under retraction is what you're looking for:
[image]
Please reopen if I've misunderstood.

closed 2021-11-20 by teachingtechYT

micromouseonline, 2021-11-20:
So it is - my mistake.
Thank you.