teachingtechYT/teachingtechYT.github.io #377: Retraction Tuning GCODE files causing printer reboot

backslashV, 2021-11-20:
Hi,
When I try to print the files generated by your tool, the printer reboots.
Model: Dremel 3D20
Slicer: Dremel DigiLab

teachingtechYT, 2021-11-21:
This is going to be very hard for me to try and replicate without owning this printer. Going to need a lot more information.
If I were in your position I would connect to the printer via terminal, and send the gcode from the file through one line at a time (copy from gcode file, paste into terminal). Soon you will get to the line that causes the reboot and I will have something to work with.
It is very likely something specific to your printer that I may not be able to fix.

backslashV, 2021-11-21:
> This is going to be very hard for me to try and replicate without owning this printer. Going to need a lot more information. If I were in your position I would connect to the printer via terminal, [...]

Thank you for your response. The printer doesn't have any wireless capabilities. The drivers for it are also discontinued, so I can't connect it to a computer either.

teachingtechYT, 2021-11-22:
Hmm. Maybe you can provide a short gcode file that work as well as the one from this site that reboots so I can compare?

backslashV, 2021-11-22:
Please find them here:
[Working](https://drive.google.com/file/d/1gY08I_TR34qdkFhIu8yga6ChanZcV9y6/view?usp=sharing)
[Not working](https://drive.google.com/file/d/1bzx8yscZ3ie51ZlCSLhEkmeXUgtVlgCb/view?usp=sharing)

teachingtechYT, 2021-11-22:
A couple of things:
1. The working file seems to position 0,0 as the centre of the bed. Try ticking that box on this site: 
[image]
In my opinion this is the most likely problem, as the printer may reset to prevent itself from running outside of the print area.
2. Your normal slicer's temperature commands don't have the tool selected. Try ticking this on this site:
[image]
3. Your gcode has `M132 X Y Z A` near the start. Never seen this before and it's not supported in Marlin. Makes sense to paste it into custom start gcode box and it will be put into the output after the homing. You also have some other lines that should go in there like `M6 T0` and `M907 X100 Y100 Z60 A100`.
4. There are multiple M204 commands through your good gcode to set different acceleration for travel vs printing moves. This site won't replicate that but it should be harmful at all.
Please have a go at these suggestions and report back.

teachingtechYT, 2021-11-28:
I'm going to assume this is stale and close it for now. Hopefully the suggestions above work.

closed 2021-11-28 by teachingtechYT

backslashV, 2021-11-28:
> I'm going to assume this is stale and close it for now. Hopefully the suggestions above work.

Sorry about the delay. Those did not work. I am changing the printer.