teachingtechYT/teachingtechYT.github.io #378: Temperature Tuning issue

mickw99, 2021-11-21:
I set my temp.  to start at 255 for petg.  when i start the prints it drops to 205 i have double checked what i can and it's set for 255 first layer.  am i over looking something i've done this before never had an issue.

mickw99, 2021-11-21:
i'm not the best with gcode but the temp settings appear to be correct. 
[temperaturesettings.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/7575659/temperaturesettings.txt)

teachingtechYT, 2021-11-21:
Need you to post the gcode too please.

mickw99, 2021-11-22:
[temperature(1).gcode.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/7582025/temperature.1.gcode.txt)

teachingtechYT, 2021-11-22:
I recently implemented a change that runs ABL and homing 50 degrees less than the target temperature but then pauses and waits for the actual target to be reached after printin. THe gcode file looks fine so I'm thinking this is probably all it is.
Please let the print run and see if the set temperature rises after homing/ABL is complete.

mickw99, 2021-11-23:
it does go back up.  guess i should have been more patient.  thanks for the help

teachingtechYT, 2021-11-23:
No problem. It was a very understandable reaction and I've added a sentence to each form to warn users.

closed 2021-11-23 by teachingtechYT