teachingtechYT/teachingtechYT.github.io #380: Incorrect origin in FlashPrint with (FlashForge Adventurer 4)

Kamholtz, 2021-12-10:
Using a print bed size of 220mm x 200mm (x,y), I get the following when I import the first layer test gcode into Flashprint:

[image]

Enabling "0,0 at centre of bed" and setting the bed diameter to 200mm results in:

[image]

I see similar results with the retraction gcode too.

teachingtechYT, 2021-12-11:
The coordinates in the gcode are the coordinates in the gcode, the site doesn't do anything tricky here.
Therefore the printer must have strange coordinates. Can you confirm where 0,0 is on this printer?
Also, I'm confused by the twin preview shown above. Which is the actual preview and what is the other one?

Kamholtz, 2021-12-12:
I'm unsure about the twin preview too. The colour of the right-most preview is what I observe when importing g-code from SuperSlicer, so I assume that's the actual print. There doesn't seem to be a way to determine what the left-most preview is (no hover tooltips or ability to inspect or click the preview).

As for the origin, it would appear that 0,0 is located at the front and centre of the bed. See the image below of the bed and the status menu showing the current coordinates:

[image]

teachingtechYT, 2021-12-12:
Thanks for the extra info. I've never seen origin point before in a 3D printer. Trying to think of a solution that doesn't clutter the interrface too much for the 99.9% of users who don't need and not too hard for me to implement. Maybe X/Y offset inputs to shift everything? That's essentially what the existing code is doing to suit different beds. I just have to make sure I don't break the base functionality in bringing this in.

teachingtechYT, 2021-12-12:
Custom offset feature added. Most peole should leave this at 0,0 but in your case you should be able to adjust the position of the gcode on the bed to suit your printer.

closed 2021-12-12 by teachingtechYT