teachingtechYT/teachingtechYT.github.io #385: Retraction Tuning settings not transferrable to PrusaSlicer

zionlion77, 2022-01-04:
Firstly, thanks for this great website and extremely useful calibration tools.

I know this type of issue has been brought up already (e.g. [#150](https://github.com/teachingtechYT/teachingtechYT.github.io/issues/150) or [#182](https://github.com/teachingtechYT/teachingtechYT.github.io/issues/182) ) but I feel like there is still no straightforward solution, yet. At least from my extensive research, I couldn't find anything helping me to translate the good retraction results I got from the website generated gcode to PrusaSlicer.

I performed a fair bit of testing in order to get the right retraction settings for my recently direct drive converted Ender 5 Pro.
[image]

However, all retraction tests except for the one directly downloaded from the TT Printer Calibration page show some more or less ugly form of stringing. 

Here's my general setup:
- Creality Ender 5 Pro
- Creality 4.2.2 mainboard with TMC2209 drivers, hence no linear advance
- direct drive conversion using Bondtech BMG, E3D V6 clone with all metal (bimetal) heatbreak with copper-plated hotend and 0.4 mm nozzle
- Hero Me Gen 5 dual fan duct using one 5015 radial fan
- various Amazon basics PLA
- PID autotuned
- flow calibrated to a multiplier of 0.93
- PrusaSlicer 2.4.0

Using the temperture tower, I get good and nearly indistinguishable results between 180 and 200 °C. Starting from 210 °C I get very little stringing in the tower. So most of my retraction tests were either at 190 or 200 °C.
Typically, the TT gcode gives good stringing results using 0.8-1 mm retraction, 40 mm/s retraction speed, no z hop or extra restart distance as in the following test and shown in the picture on the left.

```
Settings for retraction tuning form
_________________________________________________________________________

G-Code originally generated by Simplify3D(R) Version 4.1.2
This calibration test gcode modified by the Teaching Tech Calibration website: https://teachingtechyt.github.io/calibration.html
All changes are marked in the gcode with 'custom' at the end of each line. Open the gcode in a text editor and search for this to your check inputs if needed.

Nozzle diameter: 0.4 mm
Layer height: 0.2 mm

Bed: 220 x 220 mm

Base feedrate: 60 mm/sec
Perimeters: 36 mm/sec
Solid infill: 48 mm/sec
Travel moves: 100 mm/sec
First layer: 30 mm/sec

Temperatures:
Bed: 60 deg C
Hot end: 200 deg C

Part Cooling: 100% from layer 2

ABL: Restore saved ABL/manual mesh - M420 S1

Segment | Retraction distance | Retraction speed | Extra restart distance | Unretract speed | Z hop
   F    |          1.25 mm       |     40 mm/sec    |        0       mm      | 40 mm/sec       |  0 mm
   E    |          1 mm            |     40 mm/sec    |        0       mm      | 40 mm/sec       |  0 mm
   D    |          0.75 mm      |     40 mm/sec    |        0       mm      | 40 mm/sec       |  0 mm
   C    |          0.5 mm        |     40 mm/sec    |        0       mm      | 40 mm/sec       |  0 mm
   B    |          0.25 mm      |     40 mm/sec    |        0       mm      | 40 mm/sec       |  0 mm
   A    |          0 mm           |     40 mm/sec    |        0       mm      | 40 mm/sec       |  0 mm
```

[image]

The retraction test on the right side of the image shows an attempt to reproduce the test using PrusaSlicer. I used the same temperature, speeds, retraction settings, fan settings (no auto cooling), no wipe as recommended on the TT website. As you can see stringing is much worse using PS. Unfortunately, I have no access to Simplify3D to test if I can reproduce the results there. Also, I dug through both gcodes and tried to find a hint what both slicers might do differently. The only thing I could find was that PS uses relative extruder distances and the TT gcode has absolute ones, though I doubt this could explain the stringing.

Since the TT gcode repeatedly gave good retraction results with no stringing, I can basically exclude any hardware issues. Furthermore, I have tested several additional retraction settings in PS, namely various wiping settings, though none of them got rid of the stringing. I seriously have no clue what else to test at this point and would be extremely thankful for any tips what might be the decisive difference between PS genereated gcode and the TT/S3D gcode.

deendee, 2023-07-16:
Hi Zionlion, I have exactly the same issue with my Ender 3.  I have just changed to an Orbiter direct drive.  The prints are good but the string is a problem.  The TT gcode is so much better.  Did you ever find the answer to this issue?

deendee, 2023-07-17:
Hi Zionlion,
I too have given up, I spent two days on this! I did notice that the tests with the Superslicer retraction test were better than others and the Benchy was not too bad - minimal stringing.  So until I find a solution I too shall live with it!  I did read somewhere that thermal heat compound in the hot end helped - I shall try that when I have a moment.

JohnnyMo94, 2024-02-28:
I have the same problem, did you solve it?

deendee, 2024-02-29:
No, I just accepted that the prints were better in general, with a tendency to string a bit more than before. 
I did give the thermal paste a try - nothing changed! If you find a solution, please let us know, I'm sure there are others with this issue.