teachingtechYT/teachingtechYT.github.io #389: [BUG] Speed/Feedrate Tuning outputs faulty gcode

Justase, 2022-01-12:
I was trying to generate speed test tower for my printer, set parameters as needed. Since there are no Generate button, I expect Download Gcode button will output generated file. Checking thru the gcode I found feed rates do not match the requested ones. 
Please see attachments for details 
Also, on top of this bug I'd like to raise a feature request for entering travel speed manually, as automatically calculated go beyond my machine possibilities.

Thank you!

[speed.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/7852481/speed.txt)
[image]
[undefined.txt](https://github.com/teachingtechYT/teachingtechYT.github.io/files/7852506/undefined.txt)

Madmike-1337, 2022-07-25:
I would like to second this. I get the same feed rate for every layer.

AlexValCar, 2022-09-04:
I also just noticed the same problem in the gcode, but it seems that the creator of this page already forgot to fix the problems from what you can see

jk257, 2022-09-06:
+1 for this issue. At every layer in the gcode after the G92 the first G1 move has an F value that equals the "A" Section Perimeter Feedrate field. It never changes. I printed 7 identical test towers scratching my head. Additionally, I'd like to recommend that some additional explanation be added to the page regarding whether or not the M203 E value needs to be increased before running this gcode and how the perimeter feedrate translates to XY feedrate.

jk257, 2022-09-06:
Kept digging and realized he's not using the F parameter in G1 to increase the Feedrate, rather using M220 to increase the Feedrate Percentage. He mentions that in the instructions, I just missed it. If you look at the gcode in a gcode viewer and choose to view speed you should see the change in speed as change in color. My issue seems to have been not upping the extruders' max feedrate so every tower was printing at the same speed no matter what I entered in the gcode generator. Reprinting now after running M203 E250 to raise the max and test.