teachingtechYT/teachingtechYT.github.io #395: Using Ublock Origin (likely other script blockers) breaks the generated gcode

ashmandias, 2022-01-24:
If you try to generate gcode with ublock origin active, you get a file that has a NaN instead of valid numbers in many places. 
For example:

> G1 XNaN YNaN E0.0614 F1800
G1 XNaN YNaN E0.7693
G1 XNaN YNaN E0.8307
G1 XNaN YNaN E1.5386
G1 XNaN YNaN E1.6000
G1 XNaN YNaN E2.3079
G1 XNaN YNaN E2.3692
G1 XNaN YNaN E3.0772

If you take the gcode 'firstlayer' file and run it through the gcode viewer, it generates a single square, instead of the expected five -- which might be easy to overlook.

Ideally there would be a sanity check script that would throw a warning if the string NaN is found in the resulting file.

It might also make sense to add a blurb to the warning on the introduction page.

mikedaf, 2022-05-11:
made an account to say thank you, didn't even think of the blocker on my new browser.

McSneaky, 2022-05-18:
Damn, I thought site is broken and made my own firstlayer in Fusion..

Disableing all kinds of shield (adblockers) in Brave didn't work tho, had to use FireFox with all adblockers turned off

slepiavka, 2022-05-26:
Printed in incognito Chrome (I don't use Chrome so no extensions are installed) worked. Blocked Firefox yielded a bunch of `NaN`s and one square only too.

Paul-Sees-Solutions, 2022-08-08:
doesn't happen to me, firefox 102, ublock origin

PowahSerg, 2022-09-19:
I just ran into this issue and caused the mother of all clogs because I didn't check the gcode... incognito chrome fixed everything right up