teachingtechYT/teachingtechYT.github.io #402: A little more documentation on the Extruder E-Steps page

gjpc, 2022-02-24:
Thanks for a great tool! May I suggest mentioning in the text that the E-Steps/mm directive **DEFAULT_AXIS_STEPS_PER_UNIT** in Marlin _Configuration.h_ is the way to bake the new stepper constant into the Marlin firmware?  Took a while to find the correct Marlin directive once I had my new step/mm constant calibrated using **M92**.

Kithikor, 2022-03-26:
The way I am reading it, is confusing me. The SPECIAL NOTE says:
This calibration is best done with the extruder detached from the hot end...
but later:
The filament will then very slowly go through the extruder (and hot end). (after you G1 E100 F50)
Or did I miss something?

mikedaf, 2022-05-18:
It doesn't matter which way you do it, you just need to make sure the nozzle is clear and the filament flows smooth at a slow speed if you don't remove the extruder. You're only looking at how much filament gets pushed past the extruder gears.