teachingtechYT/teachingtechYT.github.io #403: What is a "Significantly Off" Measurement For Flow Calibration?

vulgerrity, 2022-03-05:
Under flow calibration it says:

"If your measurement is significantly off, the following calculator can then be used to calculate the new flow rate:"

But it does not state what "Significantly off" is.  Tenths of a millimeter? Hundredths?  How many?  I do appreciate that it says your eye is the best judge for this, but my average wall thickness is 0.424, sometimes as thick at 0.45 and as thin as 0.4.  I don't think I need to make an adjustment to my flow rate, but it would be helpful for newbies like me to know what "Significantly off" means so we have a better understanding of whether or not we should change this setting.

mikedaf, 2022-05-18:
I've had this same thought process. I put the calculated number in the slicer and it seems to work fine, but usually anywhere from 90-100 should be fine if you're measuring around .45mm. Adjust based on looks, I'm not sure what it means either.

Faulknerd, 2022-08-23:
The calibration page also confused me. My walls were coming out thicker than 0.4mm, so I used the calculator and lowered my flow rate to around 80%. This caused gaps in the bottom layers of my cube (underextrusion). I increased the flow rate in increments of 5% to see what would fix it. 

The gaps closed around 95%. This left my average wall thickness at 0.51mm. I wasn't sure if this was a problem or not so just moved on.

After some time, YouTube suggested another Teaching Tech video (https://www.youtube.com/watch?v=Mnvj6xCzikM). Michael is discussing a different calibration, but around 4:48 he works with "extrusion width". I think this might be synonymous with "wall thickness" on the calibration page.

For a 0.4mm nozzle, Simplify3D suggests an extrusion width of 0.48mm. Cura suggests a width of 0.4mm, but we may be misinterpreting the setting. Cura is the slicer I use, and with a flow rate that maintained good extrusion my average wall thickness was 0.51mm - closer to Simplify3D's value.