teachingtechYT/teachingtechYT.github.io #411: Speed & Max Flow Tuning - No Cooling Fan

PJPOO, 2022-05-03:
When generating gcode for the "Speed & Max Flow Tuning", if I choose anything but layer 2 for when the cooling fan comes on, there is no M106 generated to turn on the cooling fan.  (tried with both 0.4 nozzle and 0.6 nozzle at 0.2mm layer height settings)

bulcke, 2022-06-07:
I can confirm this, with the default (layer 3) I have no cooling

Simonlebucheron, 2022-06-27:
Same for me, no cooling when starting on layer 3, but it works when starting on layer 2.

I wanted to push the limits and I didn't understand why I can print up to 500 mm/s but with so many blobs on the decelerating parts until I found this issue and I check the printer (my mistake).
When not noticed, inspection of the prints leads to misunderstanding of the speed, acceleration and cooling basic functions.
The results where still interesting, so attached two photos of my PLA prints (with/without cooling, front and back).
[image]
[image]

tmenet, 2023-05-28:
Same for me, Cooling fan in Speed & Max Flow Tuning only works when layers are set to 2, Glad I read your readme

zerog2k, 2023-06-10:
+1
>only works when layers are set to 2

*very* frustrating, as I was pulling my hair out like others thinking I had gone crazy (can't print at 60mm/sec feedrate without it looking like crap ;)

M106 is missing from Layer 3 or Layer 5 if selected. Only present on Layer 2 if selected.

zerog2k, 2023-06-10:
looks like it's supposed to replace a token (`fan3` or `fan5`) in one of the templates in speed.js
https://github.com/teachingtechYT/teachingtechYT.github.io/blob/master/js/gcodeprocessing.js#L459-L473

but they only contain the `fan2` token, so never get added for the case of selecting layer 3 or 5