teachingtechYT/teachingtechYT.github.io #425: Results do not transfer

Degamer82, 2022-08-13:
I've been using the website custom gcode generator to calibrate a Creality Ender 3 S1.  The first layer and calibration cubes print perfectly, HOWEVER, any gcode produced from a slicer ends up printing like trash.  Some parts stick, other parts print way off the bed, almost like the leveling mesh is being ignored.  I can't print the flow calibration cube because it requires the STL to be sliced.

So far I have:

Went back and used stock cura profile, in case it was something about my settings.
Added a new printer to cura, incase there was something in my start gcode.
Swapped curas start gcode with the start gcode copy/pasted from firstlayer gcode, since that gcode seems to work.
abandoned cura all together and tried using prusa slicer.

NOTHING prints correctly except the gcode produced by the website.  What am I missing?

Faulknerd, 2022-08-21:
@Degamer82 When you use the First Layer generator, do you set "Select which method of ABL is in place" to "Restore saved ABL/manual mesh"?

The reason I ask is because that will cause the generator to output gcode which will load the saved mesh. You'll see this in the file as `M420 S1`. 

If you add this gcode to your "start gcode" in Cura (Settings > Printer > Manage Printers > Machine Settings), does it help?

Degamer82, 2022-08-25:
I have added the gcode, even the whole custom start gcode, to a fresh printer install in cura.  No improvement.  I have also noticed that the pre-installed gcode (cat, pig, etc) prints just fine.  So the issue seems to be software related, but I'm having the same issue from cura AND prussa slicer.  I'm going to try Creality slicer next and see if I still have the same issue there.

Degamer82, 2022-10-11:
I have copies that code on a fresh printer in Cura.  I have tried copying
the whole start GCODE from the teaching tech files.  Nothing seems to
work.

But, in my testing I have also found that the stock GCODE files that come
with the printer (cat, rabbit, etc) print fine.  It just seems that
everything sliced by Cura (stock profiles or my long term use ld profiles)
prints like trash.  But I also tried a file in Prusa slicer and couldn't
get that to print either.  So more experimenting seems to be required.  But
it definitely seems to be a software issue rather than a hardware issue.