teachingtechYT/teachingtechYT.github.io #431: Speed Calibration - tower code  - custom feed rate not increasing

davbardi, 2022-10-22:
Notice no matter what values set in the interactive feedrate/speed table on https://teachingtechyt.github.io/calibration.html#speed it stays at lowest.  eg for the default (20,30,40,50,60 tower) file the code generates all layers at 20 (F1200).  As shown in the gcode:
for z=9.601
   G1 X76.259 Y68.866 Z9.601 E0.0396 F1200 ; custom feedrate
   G1 X77.247 Y69.246 Z9.602 E0.0852
   G1 X78.146 Y69.481 Z9.604 E0.1253
   G1 X79.129 Y69.635 Z9.605 E0.1682
For z=48.241
    G1 X76.359 Y68.811 Z48.241 E0.0420 F1200 ; custom feedrate
    G1 X77.277 Y69.149 Z48.242 E0.0842
    G1 X78.227 Y69.387 Z48.244 E0.1264
    G1 X79.195 Y69.524 Z48.245 E0.1685

olmosbananas, 2022-10-31:
correct, took me 6 prints to realize that no matter what settings I put in, it was all the lowest. Confirmed with prusa slicer gcode viewer -- the default gcode from the site would all be a single volumetric flow rate. no change at all. 
[image]

PedroLageTavares, 2023-01-06:
Having the exact same issue. 

I'm starting at 70mm/sec:

[image]

as per Zupfe, the M220 gcode does not support decimals when setting the percentage, 

[image]

ensst6, 2023-01-10:
Same issue here. Noticed after trying speeds much higher than I've been able to print on my Ender 3 before.
Ever layer after the initial layers stays at the speed of the "A" layer.

Nixon-Joseph, 2023-01-11:
@teachingtechYT Please address - In the meantime I'll see if I can figure out how to adjust the speed manually in the gcode.

Thanks for this page! It really has helped me dial in my prints.

JanGlogr, 2023-02-19:
Hey guys. Did you solve this issue in any way? I am asking since i myself ran into this issue. I cant see any difference between having feedrate 40 or 10k (which would be funny on Ender). I love this site but i feel like this issue is in some way on each part of the calibration. I do think it has same problem in retractions .

ensst6, 2023-02-20:
Hi-
I never solved it. I just did a few trials where I assumed the baseline settings were going to apply to the entire piece. But I noticed basically no difference.  I mostly ended up with what I’ve been using for years (mostly defaults in Cura).
Agree with you that other tests also seemed fishy, tho I **thought** the retraction part actually helped me (it’s been a while, not sure).
I think this is one guy writing a tone of gcode and maybe got in over his head and didn’t have time to debug it all.
Good luck getting it to work for you.
Erik

evilDave, 2024-01-08:
I just commended on a slightly newer similar question: https://github.com/teachingtechYT/teachingtechYT.github.io/issues/477

Is everyone commenting on this issue an ender user? I have an ender 3 v2 neo running mriscoc firmware and I do not think the M220 commands work the way Michael thinks they do (and the documentation says they do to be fair), at least not always. I've noted there does seem to be a limit to how high you can set the speed (even in the LCD) and note any changes - even with ludicrously high acceleration values)

I am going to modify the gcode to remove the M220 commands and update the G1...F6000 to the correct value that I want at that point and see if each layer is different then...
What was the reasoning behind using the M220 feedrate modifier rather than explicitly setting the feedrate? I don't see a slicer using this method (even where it might make sense, eg: 'slow down for overhangs' uses the feedrate directly).

inactivist, 2024-02-21:
> What was the reasoning behind using the M220 feedrate modifier rather than explicitly setting the feedrate?

I'm a bit new to this test, but if I'm interpreting this correctly, it's to test the use of the "firmware" print rate modifier function (I'm using Klipper) which relies `M220` feature.  So this test is intended to use the same mechanism to vary the speed on each test segment.  (Someone please correct me if I'm mistaken.)

> ## The philosophy of this test - M220
> [...] This is supported in [Marlin](https://marlinfw.org/docs/gcode/M220.html), [Klipper](https://www.klipper3d.org/G-Codes.html) and [RepRapFirmware](https://duet3d.dozuki.com/Wiki/M220). Changing your feedrate mid print from the printer's LCD controls or web inteface uses this feature of the firmware. This test will insert M220 commands into the gcode for speed changes on the fly.
> M220 can be a double edge sword because it also affects the speed of retraction movements. If you are performing something like a retraction test, M220 is not suitable. In this case, however, due to the use of vase mode, M220 is a simple way for us to start with a slower feedrate and easily increase it throughout the print at set height intervals.
>
> [source](https://teachingtechyt.github.io/calibration.html#speed)