teachingtechYT/teachingtechYT.github.io #458: Suggestion: Option to add Brim to temperature tower

bgcamroux, 2023-04-02:
I am absolutely loving the comprehensive nature of the tools you have made available here. When I do the temperature tower, though, I find that once it gets to the fifth section I lose adhesion because the nozzle keeps contacting one of the sides, and at that point there's a fair amount of torque generated to pull the other end off of the print bed.

So I wonder if it would be possible to add an option to have a brim printed as well, to help increase the adhesion to the print bed.

I'm finally able to get a nice, smooth first layer, so maybe this will all change, but I think a brim might be helpful for some anyway. Also, if you do add this option, be sure to explain what a brim is so that first-time users and newbies aren't completely lost.

Thanks!

smizles, 2023-04-17:
Bump ^this. My tower rips off the bed during the bridge on the first layer. It's so frustrating, and it wouldn't be happening if I had a brim on it!

mooklepticon, 2023-09-03:
Searched for brim, found this. Agreed. Having adhesion issues after about the 3rd level.

anaklupa, 2023-09-11:
Up voting.