teachingtechYT/teachingtechYT.github.io #463: Speed & Max Flow gcode generates using only first layer Perimeter Feedrate speed

Sallowpad, 2023-04-13:
https://teachingtechyt.github.io/calibration.html#speed

As in title. Gcode Generate button only uses the Perimeter feedrate for the first layer (ie 40). This value then used for all segments A-E regardless of what user input.

hindsight351, 2023-05-08:
I have discovered the same issue.

ricsvenyus, 2025-10-20:
I have just found a similar fault in the G code output.
The feed rate for the 1st 10 mm height is correct at 50mm/s
The next band 10 to 20mm should have been 90mm/s but remained at 50mm/s.
 i.e. G1 X131.3590 Y123.8110 Z14.001 E0.0350 F3000 ; custom feedrate.
This feedrate of 50mm/s continued for whole file.
I had selected feedrates of 90, 100, 110, 120mm/s