teachingtechYT/teachingtechYT.github.io #477: Speed & Max Flow Tuning

tgnews, 2023-07-28:
I noticed that the flowrate does not change in higher layers.
I had selected the input values 20, 30, 40, 50, 60 which should result in first segment F1200, 2nd segment F1800, etc up to last segment F3600.
But the gcode only contained F1200 for all segments. Hence the model printed equally "slow" on all segments.

In addition: having segment sizes of 1 cm hight is an "overkill" and waiste of fileament. I manually modified the gcode to "cut out" the middle 5 mm per segment, leaving bottom and top of each segment intact.

would be great if you could adjust the gcode generator for this model accordingly

evilDave, 2024-01-08:
I've been having an issue with a similar thing. I was increasing the speed setting in this test and just not seeing any changes in the speed. I notice in the code (and I think this is explained at the top of the tutorial page) that the different speed settings are governed by `M220` commands to set the speed before each layer is printed with the same feedrate setting (the feedrate you specify for the first speed setting). Is there something about some printers or some setups that makes it ignore the M220 speed commands after a certain point?

I'm on an ender 3 v2 neo and was able to print a manually sliced vase model at 100mm/s and still turn the speed up to 110, 120, 130, 140 on the LCD, so was trying this test out at 100, 120, 140, 160, 180 to see what would happen... And nothing happens. The G1's for each "speed level" set the feedrate to F6000 (100 mm/s) and each level is sped up by M220 first set to S100, then S120, S140 etc. But during the print, nothing happens, nothing changes. I have acceleration turned up to 2000. Mriscoc firmware.

I am guessing something in the printer is capping the speed but I can't figure out what. From articles about getting the best speed I see advice the boils down to "keep turning up the speed and acceleration until the print fails" but I cant get it to fail, the machine just stops getting any faster... If I start the manually sliced vase at 100mm/s I can hear the speed increases up to about 140 speed (changing on the LCD or sending M220 S140 via octoprint terminal) but after that the speed doesn't increase any more.

philjmaier, 2024-01-31:
I had this issue. Running Klipper. Noticed the setting in printer.cfg was set to 300. Turns out this number is in mm/min which is 50mm/s. Increased it to 3000 and all is well.

maty87, 2025-03-14:
hi i noticed an error in the section Marlin FW section is that M203 didnt work for me with Marlin Legacy ( ithink) FW im running a VOXELAB Aquila S3 with Stock FW as far as im aware M203 just returned OK for me but M503 returned the information