teachingtechYT/teachingtechYT.github.io #482: Warping of the filament while printing "Temperature Tuning" file

SinaFarahifar, 2023-09-13:
Hi,

I am trying to print a pure PLA with Prusa i3 MK3+, but I have a problem some layers in the sample have warping which causes a crash of sample and nozzle, detaching of the sample and failure of the printing.
could you please help me?
[image]

Thanks

Wiriza01, 2023-10-03:
+1 but not sure it is code issue.  thinking cooling / material issue.. and or retraction pulling material?  Managed to use calipers to measure movement on z step which seemed fine in my case.  Still amazed at how well Michael bridge prints.  On a Sv04 similar to ref image.. maybe not so pronounced curl, but enough to hear the nozzle clip.

Duffy77, 2023-12-26:
Have the same problem. Don't know why, but it seems to be when the temperature is too low.
So when you start your temperature with higher values worked for me.
But it could also be a solution to print with descending temperatures ?
Is there a reason why it is generated ascending?

Duffy77, 2023-12-26:
Look at #349 this seems to be a cooling problem