teachingtechYT/teachingtechYT.github.io #501: Retraction test stops extruding at very start of the left tower on Prusa Mini+

risitt, 2024-10-18:
This is not necessarily a bug with your retraction tower, but I've exhausted all other explanations I could think of including printer firmware and calibration.

In short, the retraction tower fails in exactly the same way very time. I now have six identical prints of just the base of the retraction towers. The towers themselves won't print. It fails at layer five, just as it is starting the very first section of the left tower.

The failure mode is that the extruder motor no longer turns after line 2244 of the attached GCODE. It's not a clog, and it's not skipping. It just stops.

I've looked at the GCODE preview in PrusaSlicer, it looks fine? I've scrolled to the exact problem area and highlighted the GCODE which may be failing.

There is a command there (G92 E0.0000) to reset the extruder position in the printer's memory to 0. I'd say that would be a smoking gun were it not for the fact that this exact same GCODE was successfully executed 29 times prior to line 2245 when printing the base of the retraction tower.

Other notes:

- It fails both with the newest Mini+ firmware (6.1.2), as well as the prior version (6.0.4).
- I was able to print it successfully as recently as a month ago on firmware version 6.0.4.
- I've downgraded to firmware 6.0.4 and factory reset my printer, this did not help
- It's not a clog, and I've printed 500 g of stuff in the meantime
- I've tried slower print speeds for the retraction test. No help.

Photos, screenshots, problem GCODE, and retraction test settings are attached.

[image]
[image]
[image]
[retraction.gcode.txt](https://github.com/user-attachments/files/17437972/retraction.gcode.txt)
[retractionsettings.txt](https://github.com/user-attachments/files/17437996/retractionsettings.txt)

risitt, 2024-10-18:
A few more things I tried to no avail:

- Disassembled, cleaned, and reassembled extruder. There was some buildup of plastic dust in the gear. That's cleaned out now. Still the same problem.

- Replaced nozzle, done a half dozen cold pulls as I initially thought it was a clog. Nozzle was worn so it was time to replace. But clearly not the culprit here!

risitt, 2024-10-18:
Ok, so, I "solved" it. But not really in a useful way.

Removing the start GCODE for linear advance, K900 0.3, fixes the problem.

Because linear advance and retraction interact, I would very much prefer to not have to remove the linear advance setting. So this is a rather unsatisfying solution.

I suppose I will have to see if there is other start GCODE that is (for whatever reason) important, such as setting accel or jerk limits on the extruder.

risitt, 2024-10-18:
I ended up using virtual extruders in PrusaSlicer to set different retraction distances on six instances of a shape that encouraged stringing. This worked and allowed me to select a best retraction distance while linear advance was active. Still, it used more filament than a single retraction tower, and was more tedious to set up, so I still hope to be able to get your retraction test working.

I noticed that when I have that start GCODE in there, the part of the base of your retraction tower which does print really whips the extruder gears around, much moreso than the printer normally does with linear advance.

I also notice there is another bug report ( #495 ) with a similar symptom, but a different printer (Ender instead of Prusa), although it's not clear what their start GCODE was.

Related issues?

iitggithub, 2024-12-09:
Having this issue myself. I tried different retraction distances including the exact settings used by prusaslicer but still getting the same issue. I don't have the M900 command in my start gcode . Going to do the retraction test the old fashioned way.

iitggithub, 2024-12-10:
Been doing some digging....

Since your firmware is 6.1.2 which is greater than 5.0.0, you should be using M572 which replaces M900 according to 
[https://help.prusa3d.com/article/pressure-advance_814986](https://help.prusa3d.com/article/pressure-advance_814986)

Assuming you're using PLA the M572 S<value> for that filament should be 0.27:

`M572 S{if nozzle_diameter[0]==0.6}0.17{elsif nozzle_diameter[0]==0.8}0.12{elsif nozzle_diameter[0]==0.4}0.27{elsif nozzle_diameter[0]==0.25}0.85{else}0{endif}`

So M572 S0.27 is the default these days for Generic PLA and Prusament PLA. Try setting M572 S0.27 if you're tuning PLA and...

The default retraction distance on the mini with a 0.4mm nozzle is 2.5mm with a retraction speed of 70, deretraction speed of 40, 0 extra restart distance, and z hop 0.2mm.

So your retraction distance are significantly lower than the default and might even be too low for a bowden setup and your retraction speed is much lower than the default so it's not retracting as much and it's doing it slower.

I suspect that this is actually a problem that gets worse over time rather than a problem with the gcode on a specific line... and it starts from the first layer since the first 5 layers look to be using the at least be using the retraction distance settings from section A.

I think... try it with M572 S0.27 and if it still fails, try it again but with retraction distance starting from 2.5mm and retraction speed set to 70mm/s, deretraction speed set to 40mm/s, extra restart distance 0, and z hop set to 0.2mm. Print speeds should also be able to stay high too but i'm not sure what the default print speed is for a mini.

risitt, 2024-12-10:
Hi @iitggithub, you seem to be right about M572 replacing M900, although the firmware just replaces one command with the other 1:1, so it won't make a difference in the end.

Low retraction values are expected if your pressure advance is very well tuned, because pressure advance greatly decreases unwanted pressure buildup that otherwise necessitates retraction.

After weeks of calibrations I've settled on pressure advance of 0.3, and retraction distance of 0.9mm with a third party (Spool3d) PLA. The print quality is great. No stringing, almost invisible seams, sharp corners. I originally had gaps at the seams and blobby corners using Prusaslicer's defaults which is what prompted me to run this calibration.

Anyway, I never did solve this issue although I believe I've narrowed the cause down to extruder acceleration limits ending up far too high. It only occurs with pressure advance settings above 0.2 (at least for me). The Prusa default for 0.4mm nozzles, PLA on the Mini+ is 0.2, so I was able to print these retraction tests for many years before I decided to test my pressure advance and found the Prusa defaults are not so great for my third party filament.

This "bug" is probably not a bug with the teachingtech retraction test. It's really up to us to know what start GCODE we need to paste in to make our printers behave and my GCODE-fu is just not up to that task. I ended up finding a mini retraction test print (not a tower) and just printed numerous instances of it with different settings to dial in the calibration. Worked like a charm.

Personally I think this bug could be closed.